enjoy-digital/liteeth #5: If there are too many CSR registers before the Ethernet CSR registers, the softcore just reboots

mithro, 2017-01-11:
If you have too many CSR registers in the BaseSoC the NetSoC just continually reboots,

Working
```python
class BaseSoC(SoCSDRAM):
     csr_peripherals = (
         "spiflash",
         "front_panel",
         "ddrphy",
        "dna",
        "git_info",
        "platform_info",
#        "fx2_reset",
#        "fx2_hack",
#        "opsis_eeprom_i2c",
        "tofe_ctrl",
        "tofe_lsio_leds",
        "tofe_lsio_sws",
     )
```

Non-working - swapping the opsis_eeprom_i2c and tofe_ctrl things will continue to work...
```python
class BaseSoC(SoCSDRAM):
     csr_peripherals = (
         "spiflash",
         "front_panel",
         "ddrphy",
        "dna",
        "git_info",
        "platform_info",
#        "fx2_reset",
#        "fx2_hack",
        "opsis_eeprom_i2c",
        "tofe_ctrl",
        "tofe_lsio_leds",
        "tofe_lsio_sws",
     )
```

enjoy-digital, 2017-01-11:
Possible problem:
https://github.com/enjoy-digital/opsis-soc/blob/master/opsis_base.py#L222
should be:
csr_map_update(BaseSoC.csr_map, csr_peripherals)?
Can you do a test?

mithro, 2017-01-11:
Yeah, that update is correct in my branch.

mithro, 2017-01-11:
IE I still get the failure behaviour when using csr_map_update with the correct base.

enjoy-digital, 2017-01-11:
Then can you compare the two csr.h generated and see if some addresses overlap?

mithro, 2017-01-12:
I'm pretty sure this was related to libuip not being rebuilt correctly when csr.h changed.

I ended up needing the following patch;
```patch
From e4af6aa24fd39c2ba77d829af269dc1040e53c74 Mon Sep 17 00:00:00 2001
From: Tim 'mithro' Ansell <mithro@mithis.com>
Date: Thu, 12 Jan 2017 13:13:15 +1100
Subject: Fixing the Makefile deps inclusion.

diff --git a/firmware/libuip/Makefile b/firmware/libuip/Makefile
index f3c6b73..4bc93d4 100644
--- a/firmware/libuip/Makefile
+++ b/firmware/libuip/Makefile
@@ -64,7 +64,7 @@ all: $(UIPLIB)
 .PHONY: all compile clean
 
 # pull in dependency info for *existing* .o files
--include $(OBJECTS:.o=.d)
+-include $(UIPOBJS:.o=.d)
 
 %.o: %.c
 	@mkdir -p $(@D)
```

enjoy-digital, 2017-01-12:
Can you send a pull request with this?

enjoy-digital, 2017-01-12:
Closing the issue here since not related to liteeth.

closed 2017-01-12 by enjoy-digital