enjoy-digital/liteeth #10: Import the following two changes from MiSoC

mithro, 2017-10-30:
FYI These two changes landed in [misoc/cores/liteeth_mini](https://github.com/m-labs/misoc/tree/master/misoc/cores/liteeth_mini). We should make sure they get back into LiteEth....

 * https://github.com/m-labs/misoc/commit/72faa2c7fc17b2d89830bfa0ae08808d70b969e5
 * https://github.com/m-labs/misoc/commit/9d3697c3b1c6ec2c7565dd0af0dcd1f0c98d525e

rohitk-singh, 2017-11-01:
In this commit m-labs/misoc@72faa2c , previously `self.crc_error` used rising edge of `crc.error`. Do we need to do  `self.crc_error.eq(sink.eop & crc.error & ~crc_error_r)` instead to make sure of both end-of-packet as well as rising edge of crc.error also? Just wondering. @enjoy-digital @mithro

mithro, 2017-11-01:
@rohitk-singh No idea - @enjoy-digital Probably knows....

mithro, 2017-11-01:
@rohitk-singh Should also ask @whitequark who wrote the patches on the MiSoC side?

enjoy-digital, 2017-11-01:
@mithro: changes are applied. I think crc_error has been tested by @whitequark. We'll fix that if not working.

closed 2017-11-01 by enjoy-digital

whitequark, 2017-11-01:
@enjoy-digital I believe that `sink.eop` acts as a strobe for checking `crc.error`, at least liteeth is written that way elsewhere. It might still not be quite the right thing because I'm new to logic design, feel free to fix.

enjoy-digital, 2017-11-01:
@whitequark: the strobe is in fact `source.stb & source.eop`, i did the change here for liteeth: https://github.com/enjoy-digital/liteeth/commit/26c01a16277c05bd9519e85bafff761eaff4e212
you can modify misoc? (or i'll do).

whitequark, 2017-11-01:
Please do.

enjoy-digital, 2017-11-01:
Done.