enjoy-digital/liteeth #14: example_designs: invalid path

blshkv, 2018-06-01:
```
 * Detected file collision(s):
 * 
 *      /usr/lib64/python3.6/site-packages/example_designs/make.py
 *      /usr/lib64/python3.6/site-packages/example_designs/__init__.py
 *      /usr/lib64/python3.6/site-packages/example_designs/targets/__init__.py
 *      /usr/lib64/python3.5/site-packages/example_designs/make.py
 *      /usr/lib64/python3.5/site-packages/example_designs/__init__.py
 *      /usr/lib64/python3.5/site-packages/example_designs/targets/__init__.py
```

both litepcie and liteeth (and may be other lite*) install this directory into the same location from setup.py

enjoy-digital, 2018-06-01:
example_designs are not supposed to be installed. Can you try removing the __init__.py file from example designs and retry the installation?

blshkv, 2018-06-01:
Please go ahead and delete ```__init__.py```. It fixed the issue.
Thank you

enjoy-digital, 2018-07-18:
This should be solved by https://github.com/enjoy-digital/liteeth/commit/6b872fd2711905a18f15733466f711f021765dd5.

closed 2018-07-18 by enjoy-digital

blshkv, 2018-07-19:
not fixed properly:
```
 * ERROR: sci-electronics/liteeth-20180718::pentoo failed (install phase):
 *   Package installs 'test' package which is forbidden and likely a bug in the build system.

```

blshkv, 2018-07-19:
you also failed to patch all other lite* packages

enjoy-digital, 2018-07-19:
ok thanks, will have another look at that.

enjoy-digital, 2018-07-19:
it should be good now, can you test? If that's not good, can you suggest a patch since i'm not able to reproduce the issue? (i'd like to keep the __init__.py since that's useful for others things).

blshkv, 2018-07-19:
ok, it works now. Thanks!