enjoy-digital/liteeth #15: Issue with mii.py and latest litex-buildenv

mithro, 2018-08-23:
We are getting the following error;

```
./build/arty_net_lm32/output.20180823-045832.log
---------------------------------------------
Traceback (most recent call last):
  File "./make.py", line 143, in <module>
    main()
  File "./make.py", line 108, in main
    soc = SoC(platform, ident=SoC.__name__, **soc_sdram_argdict(args), **dict(args.target_option))
  File "/home/travis/build/mithro/litex-buildenv/targets/arty/net.py", line 33, in __init__
    platform.request("eth"))
  File "/home/travis/build/mithro/litex-buildenv/third_party/liteeth/liteeth/phy/__init__.py", line 21, in LiteEthPHY
    return LiteEthPHYMII(clock_pads, pads, **kwargs)
  File "/home/travis/build/mithro/litex-buildenv/third_party/liteeth/liteeth/phy/mii.py", line 94, in __init__
    self.submodules.rx = ClockDomainsRenamer("eth_rx")(LiteEthPHYMIIRX(pads))
  File "/home/travis/build/mithro/litex-buildenv/third_party/liteeth/liteeth/phy/mii.py", line 51, in __init__
    converter.reset.eq(~pads.rx_dv),
AttributeError: 'Record' object has no attribute 'rx_dv'
```

rohitk-singh, 2018-08-23:
Probably because arty platform has `dv` instead of `rx_dv` for ethernet? https://github.com/timvideos/litex-buildenv/blob/master/platforms/arty.py#L88-L98

enjoy-digital, 2018-08-23:
yes

enjoy-digital, 2018-08-23:
Closing since fixed in litex-buildenv.

closed 2018-08-23 by enjoy-digital