enjoy-digital/liteeth #16: RMII requires DDR

DurandA, 2019-02-07:
In an attempt to create a UDP source/sink for the *TinyFPGA BX*, I discovered that `LiteEthPHYRMII` requires a [`DDROutput` for tx](https://github.com/enjoy-digital/liteeth/blob/77fa4bfb1e452adb1fa34c1b0baede68c056763d/liteeth/phy/rmii.py#L102).

Since this is not the case for `LiteEthPHYMII`, is there a particular reason that RMII implementation requires DDR? My platform does not have any DDR, so this obviously trigger a NotImplementedError:
```NotImplementedError: Attempted to use a DDR output, but platform does not support them```

Is there an easy way to remove this limitation? Should I just switch to a MII PHY?

DurandA, 2019-02-07:
Awesome, thanks. I will try it and report back if I encounter any problem.

closed 2019-02-07 by DurandA

DurandA, 2019-02-07:
It works fine. I was able to export the module to Verilog.

Is it the expected behavior that we have to manually specify `special_overrides` like below?
```
verilog.convert(my_module, special_overrides=my_module.platform.toolchain.special_overrides)
```

mithro, 2019-02-07:
Just FYI, the iCE40 does have DDR I/O.