enjoy-digital/liteeth #20: LiteEthARPTable : ARP request checking

civec, 2019-09-06:
Hi all,
I'm just confused why the ARP requeste is served if `request_ip_address == cached_ip_address` .
It is part of ARPTable FSM, state "CHECK_TABLE". Couldn't then the  MAC address be invalid. This could heppened only once bacause  `request_ip_address` register is then reseted.
I would expect only the second condition to check `request.ip_address == cached_ip_address`.\
Thank you for the clarification.
```       
fsm.act("CHECK_TABLE",
            If(cached_valid,
                If(request_ip_address == cached_ip_address,
                    request_ip_address_reset.eq(1),
                    NextState("PRESENT_RESPONSE"),
                ).Elif(request.ip_address == cached_ip_address,
                    request.ready.eq(request.valid),
                    NextState("PRESENT_RESPONSE"),
                ).Else(
                    request_ip_address_update.eq(request.valid),
                    NextState("SEND_REQUEST")
                )
            ).Else(
                request_ip_address_update.eq(request.valid),
                NextState("SEND_REQUEST")
            )
        )
```

jersey99, 2020-09-10:
@civec I am not sure I understand your question very well, but hopefully the following 2 statements make sense to you.

CHECK_TABLE can be entered upon a valid new request, or upon a request timeout, or after UPDATE_TABLE (basically after a sent arp-request received a reply).

`request_ip_address` is the last request's ip_address for which the request was sent out. Now the request has come back and the cache has been updated. So time to PRESENT_RESPONSE. 

Hope that helps.

enjoy-digital, 2020-11-23:
Closing since no update and the question has been answered.

closed 2020-11-23 by enjoy-digital