enjoy-digital/liteeth #25: IPv4 total length exceeds packet length (50 bytes)

mithro, 2020-01-16:
I noticed the following error;

[image]

mithro, 2020-01-16:
[image]

jersey99, 2020-01-16:
@mithro Could you give a little more information of your setup? Has this worked for you in the past on the same device?

mithro, 2020-01-16:
This is on the arty board with https://github.com/mithro/litex-buildenv/blob/etherbone/targets/arty/etherbone.py

It uses the LiteEth UDP core.

Using the software version of LiteEth and libuip the board works fine.

enjoy-digital, 2020-01-16:
It's should be fixed with https://github.com/enjoy-digital/litex/commit/a26853702c109fb4e650ef2d53cfd631a4bec328, @mithro can you try?

mithro, 2020-01-17:
Seems to be fixed!

closed 2020-01-17 by mithro