enjoy-digital/liteeth #32: Standalone core generator does not work

ximinity, 2020-02-08:
I would like to use the LiteEth core as a standalone core within my designs. I'm using the Versa ECP5-5G board as my development platform. This repository contains a gen.py file that exists to perform exactly these steps. However, only the Xilinx-7 series of FPGAs are currently supported. I would like to add ECP5 support to this generator.

#31 is a WIP implementation for the ECP5 generator. It adds two extra flags, one to designate the desired chip and one for the platform.

The core that gets generated with the following arguments

 `./gen.py --phy rgmii --core udp --chip LFE5UM5G-45F-8BG381C --platform lattice`

synthesizes properly with trellis.

To test the core I've added the (`with_icmp=True`) option to test simple ICMP (ping) requests. However, it seems like the link doesn't get established as the LED on the Ethernet interface is still on. Ping requests also don't work. As a test, I've wired the `udp_sink_ready` signal to a LED on the Versa. This LED does turn on so it seems the core is functioning. Is there anything else I need to do at runtime for initialization (specifically for the UDP core, not Wishbone)?

I've tested the Ethernet example in the versa_ecp5 [1] repository, specifically with the following options:

`./versa_ecp5.py rgmii_test trellis`

With this test ping requests do work.

[1] https://github.com/enjoy-digital/versa_ecp5

enjoy-digital, 2020-02-12:
@ximinity thanks for the PR and feedback. Can you share your Yosys/NextPnr project here so that i can also do some testing on the Versa ECP5.

ximinity, 2020-02-22:
@enjoy-digital 

I've tested the new changes and attached the basic example that I used. I've used the following configuration for the generator:

```yml
# PHY ----------------------------------------------------------------------
phy:         LiteEthECP5PHYRGMII
vendor:      lattice
# Core ---------------------------------------------------------------------
clk_freq:    50e6
core:        udp
mac_address: 0x10e2d5000000
ip_address:  192.168.1.50
port:        6000
```
The most important difference is that I've set the `clk_freq` to 50 as it failed to meet timing closure. The example project also instantiates a PLL to generate a 50MHz clock from the 100MHz input clock.

I've modified the generator itself slightly to enable ICMP for testing, by adding `with_icmp=True` to the `LiteEthUDPIPCore`:

```diff
diff --git a/liteeth/gen.py b/liteeth/gen.py
index 37dfecb..c69bb8b 100755
--- a/liteeth/gen.py
+++ b/liteeth/gen.py
@@ -223,7 +223,11 @@ class UDPCore(PHYCore):
     def __init__(self, platform, core_config):
         PHYCore.__init__(self, core_config["phy"], core_config["clk_freq"], platform)
 
-        self.submodules.core = LiteEthUDPIPCore(self.ethphy, core_config["mac_address"], convert_ip(core_config["ip_address"]), core_config["clk_freq"])
+        self.submodules.core = LiteEthUDPIPCore(self.ethphy,
+                                                core_config["mac_address"],
+                                                convert_ip(core_config["ip_address"]),
+                                                core_config["clk_freq"],
+                                                with_icmp=True)
         udp_port = self.core.udp.crossbar.get_port(core_config["port"], 8)
         # XXX avoid manual connect
         udp_sink = self.platform.request("udp_sink")
```

The example files I tested with: [eth_example.tar.gz](https://github.com/enjoy-digital/liteeth/files/4239697/eth_example.tar.gz).

The UDP sink/source are wired directly as to try and simulate simple loopback, except that the IP address for the sink has been hardcoded to 192.168.1.100 and the port to 6000. The result is that I don't receive any UDP packets when I send them, though this might be because the wiring between sink and source isn't entirely correct. However, ping requests also fail even though the `with_icmp` option is set.

rowanG077, 2020-03-19:
@enjoy-digital Did you have a chance to test this? Currently we are blocked because of this :(.

enjoy-digital, 2020-03-19:
@rowanG077: sorry, i'll try to look at that today.

enjoy-digital, 2020-03-19:
With https://github.com/enjoy-digital/liteeth/commit/ea24ff69931699ed0d2badc7c691277ff5c8fe72 and the attached code, i'm able to ping the board. The main issue was that with 1Gbps, the system clock for now needs to be >= 125MHz. 125MHz is a bit difficult to reach on ECP5 and there are plans to improve that (https://github.com/litex-hub/litex-boards/issues/40), but for now you'll have some timing violations but it appears to work on hardware.

Can you verify it's also working for you? For testing, i just used the `eth_rx_clk` as `sys_clk` and removed the PLL, but you should be able to reintroduce it if you make sure `sys_clk_freq` is >= 125MHz.

[eth_example_2020_03_19.zip](https://github.com/enjoy-digital/liteeth/files/4356462/eth_example_2020_03_19.zip)

enjoy-digital, 2020-03-21:
@ximinity, @rowanG077: FYI this is also used here: https://github.com/NiklasFauth/colorlight-led-cube.
https://github.com/NiklasFauth/colorlight-led-cube/blob/master/fpga/liteeth.yml
https://github.com/NiklasFauth/colorlight-led-cube/blob/master/fpga/liteeth_core.v

ximinity, 2020-03-21:
@enjoy-digital 

I can confirm that the modified example at the given commit works, thanks. The project that you linked is a nice example, thanks for mentioning it! I'll close this issue as the main point has been addressed.

closed 2020-03-21 by ximinity

cbrinkerink, 2022-08-25:
My apologies for responding to a closed issue, but this seems like the best place. I am running into some very strange behaviour surrounding the ARP messages generated by the standalone LiteEth core. I have used the code in the zip-file linked in enjoy-digital's response from March 19th, 2020 to test basic UDP echo functionality:

> With [ea24ff6](https://github.com/enjoy-digital/liteeth/commit/ea24ff69931699ed0d2badc7c691277ff5c8fe72) and the attached code, i'm able to ping the board. The main issue was that with 1Gbps, the system clock for now needs to be >= 125MHz. 125MHz is a bit difficult to reach on ECP5 and there are plans to improve that ([litex-hub/litex-boards#40](https://github.com/litex-hub/litex-boards/issues/40)), but for now you'll have some timing violations but it appears to work on hardware.
> 
> Can you verify it's also working for you? For testing, i just used the `eth_rx_clk` as `sys_clk` and removed the PLL, but you should be able to reintroduce it if you make sure `sys_clk_freq` is >= 125MHz.
> 
> [eth_example_2020_03_19.zip](https://github.com/enjoy-digital/liteeth/files/4356462/eth_example_2020_03_19.zip)

I have tried both the recommended clock setup, where I use a PLL to generate a 125 MHz clock from the 100 MHz system clock, as well as the setup where the 125 MHz clock is copied from the PHY receive clock. ICMP Ping works fine, I always get immediate responses. But UDP packets only get returned after a delay of ~0.7 seconds. In the mean time, the LiteEth stack generates ARP requests to which it gets quick answers from the connected computer, but it seems to be ignoring those. Only immediately after its 8th ARP request does it decide to send the UDP echo. Here is a Wireshark screenshot showing the timing of these interactions:

[image]

In the screenshot, my host computer uses IP address 192.168.1.100 and the Versa board uses 192.168.1.50. The UDP packet transmitted at index 109 corresponds to the UDP packet received with index 125 (each UDP packet has a unique payload so I can cross-identify the sent and received packets).

Fiddling with the internal ARP timing values in the LiteEth core has some impact on the timing associated with this behaviour, but I am not sure why it occurs. Any ideas? Thanks in advance!

enjoy-digital, 2022-08-25:
@cbrinkerink: As suggested by @rowanG077, I would indeed recommend testing with current version of the code. I or @rowanG077 could probably help since we have been using it on ECP5 recently.

cbrinkerink, 2022-08-25:
Absolutely, I will be happy to try that. I did notice the `gen.py` script for the current LiteEth version no longer accepts the same flags as mentioned at the top of this issue in order to build a verilog file, so I will have to sniff around a bit to find out how to convince it to do that. If you happen to have the command-line settings to use for the ECP5 Versa board (UDP + ARP + ICMP), I would be much obliged :)

enjoy-digital, 2022-08-25:
@cbrinkerink: you can probably use this:

```
# PHY ----------------------------------------------------------------------
phy:          LiteEthECP5PHYRGMII
phy_tx_delay: 0e-9
phy_rx_delay: 2e-9
device:       LFE5U-25F-6BG256C
vendor:       lattice
toolchain:    trellis

# Core -------------------------------------------------------------------------
clk_freq:    125e6
core:        udp_raw
mac_address: 0x10e2d5000000
ip_address:  192.168.1.50
data_width:  32

# UDP Ports --------------------------------------------------------------------
udp_ports: {
  "udp0": {
    "data_width"    : 32,
  },
}
```

cbrinkerink, 2022-08-25:
Thanks for your quick responses! I have used your YML with some minor tweaks (the 'udp_raw' core entry was not recognised so I replaced it with 'udp', and I used device code LFE5UM5G-45F-8BG381C). I updated the wrapper.v code to work with the smaller number of inputs/outputs as well.

I currently get more irregular responses from the board, where multiple UDP packet payloads (I think 13 of them) seem to be combined into a single, longer, UDP response. The cadence with which ARP requests are sent out has also changed, I am now seeing one every 13 seconds.

[image]

For convenience, I attach the `udpcore.yml`, `wrapper.v` and generated `liteeth_core.v` file in a .zip below.

[udp-echo-server-25-08-2022.zip](https://github.com/enjoy-digital/liteeth/files/9426759/udp-echo-server-25-08-2022.zip)

rowanG077, 2022-08-25:
@cbrinkerink
Yeah so the "udp" interface you used has a FIFO abstraction on top of it that relinquishes control over packet size. Essentially you stream data into the core and it will look at the FIFO level and if the fifo is not empty it starts transmitting a packet with size of the FIFO level at that point in time. 

I had the same issue so @enjoy-digital made a core generator that uses the "udp_raw" interface. This gives you control over packet size. The drawback is that if you cannot source or sink data as needed you will lose data.

Here is a [gist](https://gist.github.com/rowanG077/87dddc3664cfabfa5136b00f203d2ddf) with the modified `liteeth_gen.py`.

@enjoy-digital 
Would you consider upstreaming it?

rowanG077, 2022-08-25:
@cbrinkerink
Your name was familiar so I looked it up :p. I'm actually studying a CS master at Radboud University. So if it still won't work the way you want I could bounce by your office to help tomorrow morning. I'm at Uni then anyway.

cbrinkerink, 2022-08-25:
> @cbrinkerink Yeah so the "udp" interface you used has a FIFO abstraction on top of it that partially relinquishes control over packet size. Essentially you stream data into the core and it will look at the last signal. If last is asserted it will start transmitting a packet. But that means if, for example, the first packet is very large and a few small packets follow then what happens is that all the small packets are consolidated in a single large packet.
> 
> I had the same issue so @enjoy-digital made a core generator that uses the "udp_raw" interface. This gives you control over packet size. The drawback is that if you cannot source or sink data as needed you will lose data. So to make it work accordingly you want to have FIFO between the source and sink.
> 
> @enjoy-digital Would you consider upstreaming this? It's fine by me to upstream the stuff I have commissioned to liteeth mainline.

Right, that makes sense. I suppose the FIFO is 2048 bits or so, and it transmits the aggregate packet as soon as that overflows.

If you want to drop by at the astrophysics department, you're welcome! I will be able to show you what I am doing with the board at the moment. Any time after 10AM should be okay.

Ethernet-wise, I am working towards a setup where I can have the board spit out data packets of some consistent size as defined by me, alway to the same connected computer, with no need for two-way traffic. I only need a bandwidth of ~10 Mbps or so.

cbrinkerink, 2022-08-26:
With some IRL help from @rowanG077 (thanks!), I now have a working UDP packet transmitter that transmits packets nicely on time. The trick I used to get around the FIFO buffer behaviour is to simply make the buffer 1 word-length (which is 32 bits in my case) smaller than the UDP payload size I want to send out. This causes every new packet to overflow the FIFO by 1 word, which causes it to be sent immediately. Note that in the YAML file, where the configuration for the ethernet core is defined, the fifo buffer length is stated in word lengths. To send 24-byte payloads, in the settings for the UDP port I use:

`"tx_fifo_depth" : 5,
"rx_fifo_depth" : 5`

so that the buffer is 20 bytes long (I don't actually use the rx FIFO, but I have included it here for completeness' sake).