enjoy-digital/liteeth #37: Issues with GMII

rjeschmi, 2020-03-25:
In trying to get a pano logic G2 target working I'm having some trouble with LiteEthPHYGMII and clocks management. 

I'm still quite new to litex and verilog, but am comfortable with python, so happy to run debug and logging tests at that level. 

In general I'm porting it from hdmi2usb firmware into litex-boards.

I was able to add the phy, but adding any type of core, or etherbone, starts to cause errors that look to me like naming issues with the clock domains.

The fatal error:

```
One or more errors were found during NGDBUILD.  No NGD file will be written.

Writing NGDBUILD log file "top.bld"...
Traceback (most recent call last):
  File "./pano_logic_g2.py", line 203, in <module>
    main()
  File "./pano_logic_g2.py", line 199, in main
    builder.build()
  File "/home/rob/git/litex/litex/soc/integration/builder.py", line 204, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/rob/git/litex/litex/soc/integration/soc.py", line 899, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/rob/git/litex/litex/build/xilinx/platform.py", line 48, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/rob/git/litex/litex/build/xilinx/ise.py", line 236, in build
    _run_ise(build_name, isemode, self.ngdbuild_opt, self, platform)
  File "/home/rob/git/litex/litex/build/xilinx/ise.py", line 160, in _run_ise
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

error raised:

```
ERROR:ConstraintSystem:59 - Constraint <NET "eth_rx_clk" TNM_NET =
   "PRDeth_rx_clk";> [top.ucf(80)]: NET "eth_rx_clk" not found.  Please verify
   that:
   1. The specified design element actually exists in the original design.
   2. The specified object is spelled correctly in the constraint source file.
```

I'll upload full build.log

rjeschmi, 2020-03-25:
[build-03251031.log](https://github.com/enjoy-digital/liteeth/files/4380382/build-03251031.log)

example build log

enjoy-digital, 2020-03-25:
Links to the platform and target:
https://github.com/rjeschmi/litex-boards/blob/pano_logic_g2/litex_boards/platforms/pano_logic_g2.py
https://github.com/rjeschmi/litex-boards/blob/pano_logic_g2/litex_boards/targets/pano_logic_g2.py

rjeschmi, 2020-03-25:
Also adding the ucf
[top.ucf.txt](https://github.com/enjoy-digital/liteeth/files/4380442/top.ucf.txt)

enjoy-digital, 2020-03-25:
https://github.com/enjoy-digital/liteeth/commit/fb478537e7d45512567b9b35b5a69c536cb588b2 seems to fix things with ISE. Can you try?

rjeschmi, 2020-03-25:
It does, thanks!

closed 2020-03-25 by rjeschmi