enjoy-digital/liteeth #43: High area usage on Artix 7 due to buffers infered as distributed ram

Dolu1990, 2020-07-01:
Hi,

For liteeth generated with two buffer tx and two buffer rx, i got a realy high area usage, which is made by about 1700 additional LUT as distributed ram.

Looking at the synthesis results and the generated code i got : 

```
reg [31:0] mem_4[0:381];
reg [8:0] memadr_7;
always @(posedge sys_clk) begin
end

always @(posedge sys_clk) begin
	if (ethmac_sram0_we[0])
		mem_4[ethmac_sram0_adr1][7:0] <= ethmac_sram0_dat_w[7:0];
	if (ethmac_sram0_we[1])
		mem_4[ethmac_sram0_adr1][15:8] <= ethmac_sram0_dat_w[15:8];
	if (ethmac_sram0_we[2])
		mem_4[ethmac_sram0_adr1][23:16] <= ethmac_sram0_dat_w[23:16];
	if (ethmac_sram0_we[3])
		mem_4[ethmac_sram0_adr1][31:24] <= ethmac_sram0_dat_w[31:24];
	memadr_7 <= ethmac_sram0_adr1;
end

assign ethmac_reader_memory0_dat_r = mem_4[ethmac_reader_memory0_adr];
assign ethmac_sram0_dat_r1 = mem_4[memadr_7];

reg [31:0] mem_5[0:381];
reg [8:0] memadr_8;
always @(posedge sys_clk) begin
end

always @(posedge sys_clk) begin
	if (ethmac_sram1_we[0])
		mem_5[ethmac_sram1_adr1][7:0] <= ethmac_sram1_dat_w[7:0];
	if (ethmac_sram1_we[1])
		mem_5[ethmac_sram1_adr1][15:8] <= ethmac_sram1_dat_w[15:8];
	if (ethmac_sram1_we[2])
		mem_5[ethmac_sram1_adr1][23:16] <= ethmac_sram1_dat_w[23:16];
	if (ethmac_sram1_we[3])
		mem_5[ethmac_sram1_adr1][31:24] <= ethmac_sram1_dat_w[31:24];
	memadr_8 <= ethmac_sram1_adr1;
end

assign ethmac_reader_memory1_dat_r = mem_5[ethmac_reader_memory1_adr];
assign ethmac_sram1_dat_r1 = mem_5[memadr_8];
```

```
Distributed RAM: Preliminary Mapping	Report (see note below)
+----------------------------------------------------------------------------------+----------------------------------------------+----------------+----------------------+-------------------------------------------------------------+
|Module Name                                                                       | RTL Object                                   | Inference      | Size (Depth x Width) | Primitives                                                  | 
+----------------------------------------------------------------------------------+----------------------------------------------+----------------+----------------------+-------------------------------------------------------------+
RAM128X1D x 128	 | 
|arty                                                                              | mem_4_reg                                    | Implied        | 512 x 32             | RAM16X1D x 128	RAM32X1D x 64	RAM64X1D x 64	RAM128X1D x 128	 | 
```

Even trying to force vivado picking ram blocks with attributes result into (* ram_style = "block" *) 

```
./synth_1/arty.vds:WARNING: [Synth 8-6849] Infeasible attribute ram_style = "block" set for RAM "arty/mem_4_reg",trying to implement using LUTRAM
./synth_1/arty.vds:WARNING: [Synth 8-6849] Infeasible attribute ram_style = "block" set for RAM "arty/mem_5_reg",trying to implement using LUTRAM
```

It might be related to this : 
https://github.com/enjoy-digital/liteeth/blob/master/liteeth/mac/sram.py#L225

Dolu1990, 2020-07-01:
hmmm, looking at the usage of mem_4 and mem_5, the occupancy do not addup to 1700, there might be something else.

Dolu1990, 2020-07-01:
Hoo my bad, forget my last comment, 2x (RAM16X1D x 128	RAM32X1D x 64	RAM64X1D x 64	RAM128X1D x 128) is about the ~1700 LUT, so that's realy the issue

enjoy-digital, 2020-07-13:
Thanks @Dolu1990 for the feedback/invesgation. This is fixed with https://github.com/enjoy-digital/liteeth/commit/792013a1756ea50608726ee86989ec38cfc35a8b and gives on Arty (new vs old):

[image]

cc @kgugala.

closed 2020-07-13 by enjoy-digital