enjoy-digital/liteeth #52: EtherBone data conversion issue

jedrzejboczar, 2020-11-02:
While investigating https://github.com/antmicro/litex-rowhammer-tester/issues/7 I noticed that even in simulation there are problems with data read from memory over EtherBone.

The issue is that when reading following words with expected result of e.g. `11 22 33 44 55 66 77 88` I get `11 22 33 44 55 66 77 44`. After some debugging I noticed problem in the [tx converter in udp crossbar](https://github.com/enjoy-digital/liteeth/blob/master/liteeth/core/udp.py#L55):
[image]

It seems that `sink_valid` is held high all the time, but the data is not being changed fast enough on `sink_ready`. The port writing to the `user_port` [is driven by the tx packetizer](https://github.com/enjoy-digital/liteeth/blob/master/liteeth/frontend/etherbone.py#L149).

`LiteEthEtherbonePacketTX`'s FSM is in `SEND` state all the time, `LiteEthEtherbonePacketPacketizer`'s FSM is in `ALIGNED-DATA-COPY`, so the issue must be on/before `LiteEthEtherbonePacket` sink. I will yet investigate more, but maybe you have some suggestions.

enjoy-digital, 2020-11-03:
Thanks @jedrzejboczar for the investigation. `sink_data` is indeed updated 1 cycle too late. Would you mind sharing a capture with the signals of the **packetizer** (at least sink/source of LiteEthEtherbonePacketPacketizer)? Otherwise if you have a minimal repro i can also have a closer look.

jedrzejboczar, 2020-11-03:
I checked the stream further down the path and the packetizer is ok. It goes through `Arbiter` to `LiteEthEtherboneRecord` then `LiteEthEtherboneRecordSender`: 
[image]

It seems that the problem arises from this assignment being synchronious:
https://github.com/enjoy-digital/liteeth/blob/c9334d6b3eb2060c151f1d68b2880d8bcea3c831/liteeth/frontend/etherbone.py#L314
as the data on `fifo_source` is correct but `sender_source` is too late.

jedrzejboczar, 2020-11-03:
After changing that block to `comb` it seems I am getting correct data, but I am not sure if this is entirely correct, as the `11223344` word is set on `tx_converter_sink` at previous `ready` as visible here:
[image]

jedrzejboczar, 2020-11-03:
Ah, ok, that was due to this:
https://github.com/enjoy-digital/liteeth/blob/c9334d6b3eb2060c151f1d68b2880d8bcea3c831/liteeth/frontend/etherbone.py#L323-L324
Removing line 324 fixes the problem with new data being visible during old `ready`. Do you think it should all be `comb`? Or better use `sync` and delay the `valid` signal by 1 cycle (something like changing line 321 to `source.valid.eq(~source.ready)`?

enjoy-digital, 2020-11-03:
OK good. I'll look at this to see if there was a specific reason to use `sync`, if not, we should probably indeed simplify the code to only use `comb` (and eventually add a Buffer on the `sink` or `source `if it degrades timings).

commit d96fdfc5 (2020-11-06): frontend/etherbone: simplify LiteEthEtherboneRecordSender using combinatorial logic and fix #52.
closed 2020-11-06 by enjoy-digital