enjoy-digital/liteeth #55: Problem with EtherboneRecordSender FIFO

jedrzejboczar, 2020-11-20:
I noticed one more an issue when reading data over EtherBone with burst size of 255 (buffer_depths increased accordingly).
I was getting  corrupted read data, e.g.
```
In [4]: memdump(memread(wb, 0x100))
0x40000000:  00 00 00 00 11 11 11 11 22 22 22 22 33 33 33 33  ........""""3333
0x40000010:  44 44 44 44 44 44 44 44 44 44 44 44 55 55 55 44  DDDDDDDDDDDDUUUD
0x40000020:  55 55 55 55 55 55 55 55 66 55 55 55 66 66 66 66  UUUUUUUUfUUUffff
0x40000030:  66 66 66 66 66 66 66 66 77 77 77 66 77 77 77 77  ffffffffwwwfwwww
0x40000040:  77 77 77 77 88 77 77 77 88 88 88 88 88 88 88 88  wwww.www........
0x40000050:  88 88 88 88 99 99 99 88 99 99 99 99 99 99 99 99  ................
0x40000060:  aa 99 99 99 aa aa aa aa aa aa aa aa aa aa aa aa  ................
```
The numbers should increase just as in the first line, so 0x00000000, 0x11111111, 0x22222222, 0x33333333, 0x4444444, 0x55555555 and so on. 

I run that in simulation and checked that data writter/read to/from DRAM was correct and it got corrupted in the EtherBone module. In the simulation I found that the FIFO in `LiteEthEtherboneRecordSender` was being read even when not valid. It seems that the following line makes the source valid even when FIFO is empty: https://github.com/enjoy-digital/liteeth/blob/master/liteeth/frontend/etherbone.py#L313. Changing it to `source.valid.eq(fifo.source.valid),` seems to fix the problem when testing on hardware, but for some reason it makes the simulation hang and then timeout when reading data. In the simulation the memory reads are executed and I couldn't notice any errors in the record sender, but the simulator prints `Error trying to write data to TAP device`

enjoy-digital, 2020-11-23:
Thanks @jedrzejboczar, i'm going to create benches for LiteEth similar to the one I did  recently for the others cores and I'll look at this case.

enjoy-digital, 2020-11-26:
@jedrzejboczar: this has been fixed with https://github.com/enjoy-digital/litex/commit/4a748a53b85e9f717e6c342cb0e5a2b85e6ff62d and https://github.com/enjoy-digital/liteeth/commit/d3788cd2bbf15c655c2ecbe3626dafe75606562f. The real issue here is that while doing the initial version of` etherbone.py ` I assumed the SoC was able to generate the packet faster than it was consumed by the Etherbone Sender but this is not true when using large bursts or with a slow sys_clk. So I just replaced the SyncFIFO with the new PacketFIFO that ensures the full packet is stored before starting the transmission. This adds a latency equivalent to the packet's length, but fix the issue.

closed 2020-11-26 by enjoy-digital