enjoy-digital/liteeth #60: LiteEth Linux driver: Discuss software interface, make sure it's convenient and will be stable over time.

enjoy-digital, 2021-02-09:
TODO.

fontamsoc, 2021-07-08:
Hi @enjoy-digital ,
I would be interested in helping with this as a peripheral in FontamSOC which support Wishbone4 and AXI4-Lite peripherals.
FontamSOC runs Linux; I will work on the driver and same driver can later be easily ported.
I will be using NexysA7.
Note that FontamSOC makes already use of the excellent LiteDRAM

On the NexysA7, these are the RMII PHY connections that I see support for in gen.py:
https://github.com/enjoy-digital/liteeth/blob/master/liteeth/gen.py#L73
```
##SMSC Ethernet PHY
#set_property -dict { PACKAGE_PIN C9    IOSTANDARD LVCMOS33 } [get_ports { ETH_MDC }]; #IO_L11P_T1_SRCC_16 Sch=eth_mdc
#set_property -dict { PACKAGE_PIN A9    IOSTANDARD LVCMOS33 } [get_ports { ETH_MDIO }]; #IO_L14N_T2_SRCC_16 Sch=eth_mdio
#set_property -dict { PACKAGE_PIN B3    IOSTANDARD LVCMOS33 } [get_ports { ETH_RSTN }]; #IO_L10P_T1_AD15P_35 Sch=eth_rstn
#set_property -dict { PACKAGE_PIN D9    IOSTANDARD LVCMOS33 } [get_ports { ETH_CRSDV }]; #IO_L6N_T0_VREF_16 Sch=eth_crsdv
#set_property -dict { PACKAGE_PIN C10   IOSTANDARD LVCMOS33 } [get_ports { ETH_RXERR }]; #IO_L13N_T2_MRCC_16 Sch=eth_rxerr
#set_property -dict { PACKAGE_PIN C11   IOSTANDARD LVCMOS33 } [get_ports { ETH_RXD[0] }]; #IO_L13P_T2_MRCC_16 Sch=eth_rxd[0]
#set_property -dict { PACKAGE_PIN D10   IOSTANDARD LVCMOS33 } [get_ports { ETH_RXD[1] }]; #IO_L19N_T3_VREF_16 Sch=eth_rxd[1]
#set_property -dict { PACKAGE_PIN B9    IOSTANDARD LVCMOS33 } [get_ports { ETH_TXEN }]; #IO_L11N_T1_SRCC_16 Sch=eth_txen
#set_property -dict { PACKAGE_PIN A10   IOSTANDARD LVCMOS33 } [get_ports { ETH_TXD[0] }]; #IO_L14P_T2_SRCC_16 Sch=eth_txd[0]
#set_property -dict { PACKAGE_PIN A8    IOSTANDARD LVCMOS33 } [get_ports { ETH_TXD[1] }]; #IO_L12N_T1_MRCC_16 Sch=eth_txd[1]
#set_property -dict { PACKAGE_PIN D5    IOSTANDARD LVCMOS33 } [get_ports { ETH_REFCLK }]; #IO_L11P_T1_SRCC_35 Sch=eth_refclk
#set_property -dict { PACKAGE_PIN B8    IOSTANDARD LVCMOS33 } [get_ports { ETH_INTN }]; #IO_L12P_T1_MRCC_16 Sch=eth_intn
```

How does SOC communicates with LiteEth ? Is it a stream interface that receives TCP or UDP packets ?

enjoy-digital, 2021-07-08:
Hi @fontamsoc,

very nice project work on FontamSoC, you really did quite a bit of work!

For the LiteEth integration in standalone mode (as I think you are using LiteDRAM in FontamSoC), the interface is a wishbone interface, that gives accesses to the write/read buffers and to the registers. The Microwatt project already uses LiteEth in standalone mode, so you can find useful examples here:
- https://github.com/antonblanchard/microwatt/blob/master/liteeth/gen-src/arty.yml
- https://github.com/antonblanchard/microwatt/blob/master/liteeth/gen-src/generate.sh
- https://github.com/antonblanchard/microwatt/blob/master/liteeth/generated/arty/liteeth_core.v

For the Linux driver, you can find it here:
- https://github.com/litex-hub/linux/tree/litex-rebase

Happy to help you if you have questions while studying this.

Florent