enjoy-digital/liteeth #63: very large LUT count on intel cyclone

mczerski, 2021-02-26:
Hi,

I'm trying to use liteeth with de0nano which has cyclone IV FPGA, but the synthesis generates like 60k+ LUTs just for liteeth core. It seems for me that this is way to much of logic elements. What area usage I should expect from this core ? Is this an issue with intel cyclone FPGA's only ?

mczerski, 2021-03-05:
I think I found the problem. In LiteEthMACWishboneInterface there are two frame buffers, wb_rx_sram_ifs and wb_tx_sram_ifs. the second one (wb_tx_sram_ifs) is created with read_only=False (so CPU can write transmit frames). This transmit buffer generates this verilog code:
```
always @(posedge sys_clk) begin
    if (main_sram0_we[0])
      mem_5[main_sram0_adr1][7:0] <= main_sram0_dat_w[7:0];
    if (main_sram0_we[1])
      mem_5[main_sram0_adr1][15:8] <= main_sram0_dat_w[15:8];
    if (main_sram0_we[2])
      mem_5[main_sram0_adr1][23:16] <= main_sram0_dat_w[23:16];
    if (main_sram0_we[3])
      mem_5[main_sram0_adr1][31:24] <= main_sram0_dat_w[31:24];
    memadr_10 <= main_sram0_adr1;
end
```
I think that quartus have problem with such construction and does not infer ram correctly.
If I change the code like this:
```
always @(posedge sys_clk) begin
    if (|main_sram0_we)
        mem_5[main_sram0_adr1] <= main_sram0_dat_w;
    //if (main_sram0_we[0])
    //  mem_5[main_sram0_adr1][7:0] <= main_sram0_dat_w[7:0];
    //if (main_sram0_we[1])
    //  mem_5[main_sram0_adr1][15:8] <= main_sram0_dat_w[15:8];
    //if (main_sram0_we[2])
    //  mem_5[main_sram0_adr1][23:16] <= main_sram0_dat_w[23:16];
    //if (main_sram0_we[3])
    //  mem_5[main_sram0_adr1][31:24] <= main_sram0_dat_w[31:24];
    memadr_10 <= main_sram0_adr1;
end
```
the memory is inferred correctly and project fits into FPGA.

mczerski, 2021-03-05:
and this version should be analogous to the original but makes quartus happy:
```
integer i_mem_5;
always @(posedge sys_clk) begin
    for (i_mem_5=0; i_mem_5<4; i_mem_5=i_mem_5+1) begin
        if (main_sram0_we[i_mem_5])
            mem_5[main_sram0_adr1][i_mem_5*8 +: 8] <= main_sram0_dat_w[i_mem_5*8 +: 8];
    end
    //if (main_sram0_we[0])
    //  mem_5[main_sram0_adr1][7:0] <= main_sram0_dat_w[7:0];
    //if (main_sram0_we[1])
    //  mem_5[main_sram0_adr1][15:8] <= main_sram0_dat_w[15:8];
    //if (main_sram0_we[2])
    //  mem_5[main_sram0_adr1][23:16] <= main_sram0_dat_w[23:16];
    //if (main_sram0_we[3])
    //  mem_5[main_sram0_adr1][31:24] <= main_sram0_dat_w[31:24];
    memadr_10 <= main_sram0_adr1;
end
```

rowanG077, 2021-03-05:
I would like to keep this open. This is still an issue. The code generated should be inferred to block rams on all platforms.

mczerski, 2021-03-05:
Below is an equivalent verilog code of original, but is correctly inferring ram memory by quartus:
```
always @(posedge sys_clk) begin
  if (main_sram0_we[0])
      mem_5_0[main_sram0_adr1] <= main_sram0_dat_w[7:0];
end
always @(posedge sys_clk) begin
  if (main_sram0_we[1])
      mem_5_1[main_sram0_adr1] <= main_sram0_dat_w[15:8];
end
always @(posedge sys_clk) begin
  if (main_sram0_we[2])
      mem_5_2[main_sram0_adr1] <= main_sram0_dat_w[23:16];
end
always @(posedge sys_clk) begin
  if (main_sram0_we[3])
      mem_5_3[main_sram0_adr1] <= main_sram0_dat_w[31:24];
end
```
Seems like quartus is not happy with byte write enables on single 32bit memory. I had similar issue [here](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/202), but there is actualy a byte write enable and somehow quartus is able to infere it properly.

The other solution would be to not use byte write enables. It seems for me that byte write enable is not necessary for tx/rx ethernet buffer. Potentially only last piece of frame may be not aligned to 32bit words, but that may be handled by the driver. This solution would waste less memory, because, at least on cyclone IV, for each byte write enable there needs to be separate M9K memry block (which holds 9k bits), so for single 1530 byte packet there needs to be 4 M9K blocks used, and without byte write enables only 2 M9K blocks are needed.

mczerski, 2021-03-09:
The comment about separate M9K block for each byteenable signal is not true. M9K blocks support byteenables, and this is the code that works as expected (the key here is the delaration of the mem_5 reg) :
```
reg [3:0][7:0] mem_5[0:381];
reg [8:0] memadr_9;
reg [8:0] memadr_10;
always @(posedge sys_clk) begin
    memadr_9 <= main_ethmac_reader_memory0_adr;
end

always @(posedge sys_clk) begin
    if (main_ethmac_sram0_we[0])
        mem_5[main_ethmac_sram0_adr1][0] <= main_ethmac_sram0_dat_w[7:0];
    if (main_ethmac_sram0_we[1])
        mem_5[main_ethmac_sram0_adr1][1] <= main_ethmac_sram0_dat_w[15:8];
    if (main_ethmac_sram0_we[2])
        mem_5[main_ethmac_sram0_adr1][2] <= main_ethmac_sram0_dat_w[23:16];
    if (main_ethmac_sram0_we[3])
        mem_5[main_ethmac_sram0_adr1][3] <= main_ethmac_sram0_dat_w[31:24];
    memadr_10 <= main_ethmac_sram0_adr1;
end

assign main_ethmac_reader_memory0_dat_r = mem_5[memadr_9];
assign main_ethmac_sram0_dat_r1 = mem_5[memadr_10];
```

enjoy-digital, 2021-03-11:
Thanks @mczerski for reporting and looking at this. 

Generating generic verilog that infers block ram correctly for the different FPGA families is indeed complicated. In the long term we should probably have specific implementation for the different devices, but use of `FullMemoryWE` should provide a temporary workaround (This just splits the memories in simple ones that the synthesis tools generally have less difficulties inferring as block ram) : https://github.com/enjoy-digital/liteeth/commit/6c3af746e28f55089c9f79966205499af394ae34. Can you test this? (I already did a test on an Intel device with Ethernet).

mczerski, 2021-03-11:
@enjoy-digital I will try this. This solution will increase the memory usage 2x because splitting those bytes into separate always blocks makes quartus to infer one M9K ram block for each byte (as I mentioned ealrier). FullMemoryWE is used at least in one other place in litex and it causes large memory waste due to allocating small memory blocks - my project for de0nano uses 60% of all available bits but 100% of available M9K block.

enjoy-digital, 2021-03-11:
@mczerski: Thanks for the feedback, as stated this is a temporary solution until someone find time to implement a better one :)
For this we will need to create specialized memory modules in LiteX.

Here are the results I had when testing on the c10lprefkit:

[image]

mczerski, 2021-03-11:
I tested the fix and it works as expected

closed 2021-03-12 by mczerski