enjoy-digital/liteeth #66: Use LiteEth to add ethernet to an ECP5 with a PMOD

cjearls, 2021-04-17:
Hello,

I was wondering if LiteEth currently supports any PMODs, and if not, what the best way to add a LiteEth ethernet interface to an ECP5 with 25k LUTs that doesn't have an ethernet jack on the board would be.

enjoy-digital, 2021-04-22:
Hi @cjearls,

the easiest way through a PMOD would be a RMII interface. It seems this cheap module that is also used on Arduino/ESP2 projects could be used: https://www.waveshare.com/lan8720-eth-board.htm You would just have to rotate it by 180°.

To use this with LiteX, you would just need to add an extension to your platform, for example for an SDCard: 
```python3
_sdcard_pmod_ios = [
    ("sdcard_pmoda", 0,
        Subsignal("clk",  Pins("pmoda:3")),
        Subsignal("cmd",  Pins("pmoda:1"), Misc("PULLMODE=UP")),
        Subsignal("data", Pins("pmoda:2 pmoda:4 pmoda:5 pmoda:0"), Misc("PULLMODE=UP")),
        Misc("SLEWRATE=FAST"),
        IOStandard("LVCMOS33"),
     )
]
self.platform.add_extension(_sdcard_pmod_ios)
```

And define the IOs similarly to  the other RMII PHYs (here from NeTV2):
```python3
    # RMII Ethernet
    ("eth_clocks", 0,
        Subsignal("ref_clk", Pins("D17")),
        IOStandard("LVCMOS33"),
    ),
    ("eth", 0,
        Subsignal("rst_n",   Pins("F16")),
        Subsignal("rx_data", Pins("A20 B18")),
        Subsignal("crs_dv",  Pins("C20")),
        Subsignal("tx_en",   Pins("A19")),
        Subsignal("tx_data", Pins("C18 C19")),
        Subsignal("mdc",     Pins("F14")),
        Subsignal("mdio",    Pins("F13")),
        Subsignal("rx_er",   Pins("B20")),
        Subsignal("int_n",   Pins("D21")),
        IOStandard("LVCMOS33")
     ),
```
Since the number of IOs is limited on the PMOD, some optional IOs will probably not be present (maybe `rst_n`, `rx_er`, `int_n`).

And integrate it like this in your design:
```python3
        # Ethernet ---------------------------------------------------------------------------------
        from liteeth.phy.rmii import LiteEthPHYRMII
        self.submodules.ethphy = LiteEthPHYRMII(
                clock_pads = self.platform.request("eth_clocks"),
                pads       = self.platform.request("eth"))
        self.add_ethernet(phy=self.ethphy)
```

enjoy-digital, 2021-06-11:
@cjearls: This  solution has been tested with a cheap LAN8720 module on an Arty S7 and just requires a minor modification to the LAN8720 module:

[image]
[image]

Integrated in the LiteX SoC with the following code:

```python3

        # Ethernet ---------------------------------------------------------------------------------

        # Add IOs to Board's platform.

        from litex.build.generic_platform import Pins, Subsignal, IOStandard
        def eth_lan8720_rmii_pmod_io(pmod):
            # Lan8020 RMII PHY "PMOD": To be used as a PMOD, MDIO should be disconnected and TX1
            # connected to PMOD8 IO.
            return [

                ("eth_clocks", 0,
                    Subsignal("ref_clk", Pins(f"{pmod}:6")),
                    IOStandard("LVCMOS33"),
                ),
                ("eth", 0,
                    Subsignal("rx_data", Pins(f"{pmod}:5 {pmod}:1")),
                    Subsignal("crs_dv",  Pins(f"{pmod}:2")),
                    Subsignal("tx_en",   Pins(f"{pmod}:4")),
                    Subsignal("tx_data", Pins(f"{pmod}:0 {pmod}:7")),
                    IOStandard("LVCMOS33")
                 ),
            ]
        platform.add_extension(eth_lan8720_rmii_pmod_io("pmodb"))

        # Create RMII Ethernet PHY.
        from liteeth.phy.rmii import LiteEthPHYRMII
        self.submodules.ethphy = LiteEthPHYRMII(
            clock_pads = platform.request("eth_clocks"),
            pads       = platform.request("eth"),
            refclk_cd  = None)

        # Add Ethernet MAC and connect it to the Soft-CPU.
        #self.add_ethernet(phy=self.ethphy)

        # Add Etherbone and connect if to the SoC.
        self.add_etherbone(phy=self.ethphy)
```