enjoy-digital/liteeth #70: Too much blockram usage on ecp5 (colorlight 5a-75b)/Excessive blockram usage

david-sawatzke, 2021-06-19:
The colorlight target doesn't build anymore, due to too much blockram being used. It seems like tx/rx slots use up an excessive amount.

## Versions
Yosys 0.9+4081 (git sha1 5a73f296c, gcc 11.1.0 -march=native -mtune=generic -O2 -fno-plt -fPIC -Os)
nextpnr-ecp5 -- Next Generation Place and Route (Version c73d4cf6)
Project Trellis ecppack Version 1.0-505-g0e6a320
(current git versions)
## Without
```sh
$ ./colorlight_5a_75x.py --build --with-ethernet
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:  4064/12144    33%
Info: 	          TRELLIS_IO:    51/  197    25%
Info: 	                DCCA:     2/   56     3%
Info: 	              DP16KD:    41/   56    73%
Info: 	          MULT18X18D:     4/   28    14%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     1/    2    50%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:     4/  128     3%
Info: 	            SIOLOGIC:    44/   69    63%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

## With Ethernet
```sh
$ ./colorlight_5a_75x.py --build --with-ethernet
```

```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:  5400/12144    44%
Info: 	          TRELLIS_IO:    66/  197    33%
Info: 	                DCCA:     3/   56     5%
Info: 	              DP16KD:    69/   56   123%
Info: 	          MULT18X18D:     4/   28    14%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     1/    2    50%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:    15/  128    11%
Info: 	            SIOLOGIC:    44/   69    63%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

## With Ethernet (ntxslot = 1)

```sh
$ ./colorlight_5a_75x.py --build --with-ethernet
```

```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:  5339/12144    43%
Info: 	          TRELLIS_IO:    66/  197    33%
Info: 	                DCCA:     3/   56     5%
Info: 	              DP16KD:    61/   56   108%
Info: 	          MULT18X18D:     4/   28    14%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     1/    2    50%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:    15/  128    11%
Info: 	            SIOLOGIC:    44/   69    63%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

## With Ethernet (ntxslot = 1, nrxslot = 1)

```sh
$ ./colorlight_5a_75x.py --build --with-ethernet
```

```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:  5306/12144    43%
Info: 	          TRELLIS_IO:    66/  197    33%
Info: 	                DCCA:     3/   56     5%
Info: 	              DP16KD:    57/   56   101%
Info: 	          MULT18X18D:     4/   28    14%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     1/    2    50%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:    15/  128    11%
Info: 	            SIOLOGIC:    44/   69    63%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

## With Ethernet (ntxslot = 0, nrxslot = 1)

```sh
$ ./colorlight_5a_75x.py --build --with-ethernet
```

```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:  5214/12144    42%
Info: 	          TRELLIS_IO:    66/  197    33%
Info: 	                DCCA:     3/   56     5%
Info: 	              DP16KD:    49/   56    87%
Info: 	          MULT18X18D:     4/   28    14%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     1/    2    50%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:    15/  128    11%
Info: 	            SIOLOGIC:    44/   69    63%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```
## With Ethernet (ntxslot = 4, nrxslot = 1)

```sh
$ ./colorlight_5a_75x.py --build --with-ethernet
```

```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:  5403/12144    44%
Info: 	          TRELLIS_IO:    66/  197    33%
Info: 	                DCCA:     3/   56     5%
Info: 	              DP16KD:    81/   56   144%
Info: 	          MULT18X18D:     4/   28    14%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     1/    2    50%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:    15/  128    11%
Info: 	            SIOLOGIC:    44/   69    63%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

It seems that a txslot takes 8 dp16kd (naive calculation means a tx slot takes 16kB, that's probably not how they're used) and an rxslot takes 4 dp16kd.

In my (very naive) search for ntxslots memory usage I only found https://github.com/enjoy-digital/liteeth/blob/6febb1aaabfb8ad83bc1c4de2d366410dae63ed5/liteeth/mac/sram.py#L277 which should only need one blockram, so I'm not quite sure where the other ones come from.

david-sawatzke, 2021-08-03:
I've investigated this a bit more: The memory get has a port with a we granularity of 8, which means that `FullMemoryWE` transforms the buffer into 4 grains, so 4 different memory instances in the generated verilog. Yosys probably had a regression and couldn't catch this anymore.

But today I updated yosys and it seems to be fixed now:
## With Ethernet
```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:  5466/12144    45%
Info: 	          TRELLIS_IO:    65/  197    32%
Info: 	                DCCA:     3/   56     5%
Info: 	              DP16KD:    34/   56    60%
Info: 	          MULT18X18D:     4/   28    14%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     1/    2    50%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:    15/  128    11%
Info: 	            SIOLOGIC:    44/   69    63%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     1/    1   100%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

Based on that, I'll go ahead and close this issue

rowanG077, 2021-08-04:
For reference I don't have this problem. But I use the standalone core generator.

mithro, 2021-08-04:
@umarcor - You were talking about this in reference to GHDL or something?

Disasm, 2021-08-08:
I found the commit: https://github.com/enjoy-digital/liteeth/commit/6c3af746e28f55089c9f79966205499af394ae34
Before: `DP16KD:    51/   56    91%`
After: `DP16KD:    69/   56   123%`

david-sawatzke, 2021-08-08:
For some reason github didn't pick it up, but I opened a PR to make this configurable: https://github.com/enjoy-digital/liteeth/pull/72

umarcor, 2021-08-09:
> @umarcor - You were talking about this in reference to GHDL or something?

Yes. @stnolting wants to use "tool agnostic VHDL" in [NEORV32](https://github.com/stnolting/neorv32), so we did several tests for inferring memories with GHDL and Yosys. There were two main issues: global enable and byte granularity.

- https://github.com/ghdl/ghdl/issues/1780
- https://github.com/ghdl/ghdl/issues/1781
- https://github.com/ghdl/ghdl/issues/1782

commit a16bfdfc (2021-08-11): Merge pull request #72 from david-sawatzke/fullmemwe

mac: Allow configuring usage of FullMemoryWE (fixes #70)
commit e14c90db (2021-08-11): mac: Allow configuring usage of FullMemoryWE (fixes #70)

On ecp5 `FullMemoryWE` leads to an increase of DP16KD block mem, while
it works better on Intel/Altera devices according to
6c3af746e28f55089c9f79966205499af394ae34.

Simple solution: Make it configurable
closed 2021-08-11 by enjoy-digital