enjoy-digital/liteeth #89: CI/Doc: Add tests for 32-bit/64-bit Datapath.

enjoy-digital, 2021-11-01:
@lschuermann, @david-sawatzke: Thanks a lot for the work on the 32-bit/64-bit Datapath. Would you mind listing here the tests/commands you are currently running to verify your changes (at least in simulation)? This will be useful now for the PRs reviews/eventual modifications and will be useful for later to document these and add CI. Thanks!

david-sawatzke, 2021-11-01:
I have used a couple different approaches depending on the issue at hand:
- ARP works by just resolving the IP (tests the MAC, hybrid mac and hardware stack)
- tftp works to copy data from the bios (variable length data gets received via `sram.py`)
- ping with varying lengths to exercise the `last_be` options and verify the Packetizer/Depacketizer works with different `last_be`s
  - The icmp checksum fails if data is missing
- etherbone to read data, primarily with the crossbar uart to check the data is valid.

Usually coupled with wireshark and the debug prints in the bios to take a look at the packets.

I've also tested the MAC & SRAM stack on the colorlight 5a-75b, with the bios and firmware (smoltcp) that handles (hacky) telnet and some art-net udp.

For #88 we mostly worked with `ping`.

The simulation is a bit hacky, but I usually test with `./litex_sim.py   --with-sdram --ethernet-phy-model=gmii  --with-etherbone --with-ethernet` and adjusting the code in litex_sim to test the correct thing. For #88 I mostly used https://gist.github.com/david-sawatzke/d0463060d84ce1748370c0888b1f6343.

enjoy-digital, 2021-11-01:
Thanks @david-sawatzke, I indeed mostly wanted to know the used commands to also be able to run similar simulations.