enjoy-digital/liteeth #101: Broadcast reception?

sergachev, 2022-01-25:
Because of this filtering by target IP https://github.com/enjoy-digital/liteeth/blob/master/liteeth/core/ip.py#L212 LiteEth devices do not receive broadcast packets - ones sent to .255 etc addresses.
In my project I just removed this comparison and do not see drawbacks yet, but not sure it's the best way.

enjoy-digital, 2022-01-28:
Thanks for the feedback @sergachev, I'll have a look at if and see if there was a valid reason for it.

david-sawatzke, 2022-01-30:
I can see multiple reasons:
- Often (not always) it's not intended for the gateware to receive broadcast packets
- If no IP check is performed, there may be instances where a packet intended for a different IP is received (especially with the new changeable IP feature). I think all packets are filtered by mac beforehand so this is an edge case
- Using only parts of LiteETH in an external project, where (for example) there might not be any mac filtering

sergachev, 2022-01-30:
> * Often (not always) it's not intended for the gateware to receive broadcast packets

Then it's worth making such filtering an option or at least documenting that it's present so that people expecting broadcast to work save time searching for reasons why it does not.

> * If no IP check is performed, there may be instances where a packet intended for a different IP is received (especially with the new changeable IP feature). 

How? Any modern switch will forward to a device only traffic intended for it.

> I think all packets are filtered by mac beforehand so this is an edge case

Not sure, because broadcast addressed to FF...FF mac gets received fine just by removing the filtering by IP that I linked above.

> * Using only parts of LiteETH in an external project, where (for example) there might not be any mac filtering

Another way can be to forward destination IP to next layers to let them decide if a packet needs processing.

enjoy-digital, 2022-02-16:
RX Broadcast is now enabled by default with https://github.com/enjoy-digital/liteeth/commit/aff1916a039815861f428270e48fe0e4175cc785. A parameter has been added to enabled/disable it and should ease code comprehension.

TX Broadcast is now also possible with https://github.com/enjoy-digital/liteeth/commit/dc96269c8ffce46b6ed61ac85a3c6bdc74946692.

closed 2022-02-16 by enjoy-digital