enjoy-digital/liteeth #103: Request for info about daisy-chaining 5a75b boards with etherbone

faeboli, 2022-02-25:
Hello,
I'm playing with colorlight 5A75B board, with a etherbone client on it.
Now it's connected directly to pc ethernet port trough phy1 connector.
Would it be possible to daisy chain another similar board, with the same firmware
(or fairly similar, changing what is needed, eg the ip address ecc)
to phi connector 2 and be able to access it from the same pc ?
PC<>----<>phi1-board1-phi2<>-------<>phi1-board2-phi2
I'd like to avoid using a network switch
Would it be hard to implement?
Is there any example of this topology?
Thanks!

enjoy-digital, 2022-02-27:
Hi @faeboli,

that's indeed an interesting feature I'd also like to explore. This will however not be directly supported by LiteEth and will requires some modifications on the MAC. The first thing would first be to validate the second PHY in the Colorlight and then think about the MAC adaptations required to redirect traffic not destined to the current board the daisy chained one.

faeboli, 2022-02-27:
@enjoy-digital  thanks for your answer
Lets's fave the first step: can I help validating the second phy?
Should I try to modify my code so that the connection uses phy2 instrad of phy1? Would that help?
I'm very interested in this feature for my application.
My usecase is for a ethernet based machine control, and avoiding  an ethernet switch would enhance the system robustness.

enjoy-digital, 2022-02-28:
Hi @enjoy-digital,

validating the second PHY would indeed be an interesting first step. The second PHY can be selected  with `--eth-phy=0` [here](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/colorlight_5a_75x.py#L192).

IIRC I did a quick test in the past but it wasn't working. I was not sure if it was related to an issue in the reversed pinout or something else. If you want to go further, you can try to debug with LiteX BIOS netboot and` software debug=True` on ethernet integration: https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1453.

When running the netboot, it will also display the TX/RX frames, giving you more visibility and ability to compare with Wireshark captures.

Maybe you could first setup this test with `--eth-phy=0` to have a reference and then switch to `--eth-phy=1`.

faeboli, 2022-02-28:
Just made one quick test:
1- build command ./colorlight_5a_75x.py --revision=8.0 --eth-phy=0 --with-etherbone --build
loaded and board answers to ping only on connector 1 (phy 0) and not on connector 2 (phy 1)
2- build command ./colorlight_5a_75x.py --revision=8.0 --eth-phy=1 --with-etherbone --build
loaded and board answers to ping only on connector 2 (phy 1) and not on connector 1 (phy 0)

enjoy-digital, 2022-02-28:
@faeboli: OK good if it works, IIRC I was doing the test with a board of revision 7.0. Now we would need to think about the best way to daisy chain the board and how to modify the MAC. I'll try to think a bit about it. Feel also free to explore the code and share your thoughts here.

faeboli, 2022-02-28:
Tested also the crossover uart on etherbone and I confirm that it's fully working on phy1.
I'll go trough the code but my knowledge about ethernet coding is very small, so probably my contribution will be not very much.
Thank you for your help on this!

enjoy-digital, 2022-02-28:
@faeboli: This is a use case I'm interested in but can't do a context switch for now to think about it more. I'll try to do it soon. This could create some funny FPGA setups.