enjoy-digital/liteeth #114: Where are the Wishbone wires?

shingarov, 2022-06-19:
I must be missing something truly elementary.  When I run `gen.py` with something like the following yml:
```
phy:        LiteEthPHYMII
vendor:     xilinx
clk_freq:   100e6
core:       wishbone
endianness: little
ntxslots:   2
nrxslots:   2
soc:
    mem_map:
        ethmac: 0x00010000
```
the module in the generated liteeth_core.v only has the MII wires:
```
module liteeth_core (
    input  wire sys_clock,
    input  wire sys_reset,
    input  wire mii_eth_clocks_tx,
    input  wire mii_eth_clocks_rx,
    output wire mii_eth_rst_n,
    inout  wire mii_eth_mdio,
    output wire mii_eth_mdc,
    input  wire mii_eth_rx_dv,
    input  wire mii_eth_rx_er,
    input  wire [3:0] mii_eth_rx_data,
    output reg  mii_eth_tx_en,
    output reg  [3:0] mii_eth_tx_data,
    input  wire mii_eth_col,
    input  wire mii_eth_crs,
    output wire interrupt
);
```
How can it work?  Have I missed something that is required for the wishbone wires to be there?  Or is there some magic that I am missing allowing LiteEth to communicate on the wishbone without having `wishbone_adr` etc in the module's interface?

Xiretza, 2022-06-25:
This was probably broken by 8733aecf89d56e1215dedd491bcd8fdea3fb21d9, or a later change in litex itself.

enjoy-digital, 2022-06-27:
Thanks, this is fixed with https://github.com/enjoy-digital/liteeth/commit/f192183255adb527614770394fa863af98f28543.

closed 2022-06-27 by enjoy-digital