enjoy-digital/liteeth #123: subtle difference between `s7rgmii` and `ecp5rgmii` phy implementations?

gsomlo, 2022-11-30:
When using bare-bones busybox-based (64-bit, on RocketChip) Linux, ethernet seems to work the same on both `digilent_nexys_video` and `lambdaconcept_ecpix5`. I can bring up `eth0`, run `udhcpc` to get a dhcp lease, manually bring the interface up, etc. Everything seems OK.

However, if I boot Fedora with (enabled-by-default) `NetworkManager`, the `ecp5rgmii` PHY behaves differently (worse) than `s7rgmii`.

Here are my build command lines:

  - On `lambdaconcept_ecpix5` (using yosys/trellis/nextpnr):

  ```
  litex-boards/litex_boards/targets/^Cmbdaconcept_ecpix5.py --build \
      --yosys-flow3 \
      --cpu-type rocket --cpu-variant fulld --sys-clk-freq 50e6 \
      --with-ethernet --with-sdcard
  ```

  - On `digilent_nexys_video` (using Vivado):

  ```
  litex-boards/litex_boards/targets/digilent_nexys_video.py --build \
      --cpu-type rocket --cpu-variant fulld --sys-clk-freq 50e6 \
      --with-ethernet --with-sdcard
  ```

On the `nexys_video`, once fedora is booted up and I log in as root, I see the following (expected) results:

```
$ dmesg | grep eth0
[  135.770864] liteeth 12001000.mac eth0: irq 2 slots: tx 2 rx 2 size 2048

$ nmcli
eth0: connected to Wired connection 1
        "eth0"
        ethernet (liteeth), D2:42:0F:6C:B6:E2, hw, mtu 1500
        ip4 default
        inet4 192.168.2.156/24
        route4 0.0.0.0/0
        route4 192.168.2.0/24
        inet6 fe80::bd84:14d8:d957:ce75/64
        route6 fe80::/64
        route6 ff00::/8

lo: unmanaged
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

DNS configuration:
        servers: 192.168.2.1
        domains: lan
        interface: eth0

Use "nmcli device show" to get complete information about known devices and
"nmcli connection show" to get an overview on active connection profiles.

Consult nmcli(1) and nmcli-examples(7) manual pages for complete usage details.

$ systemctl status NetworkManager
● NetworkManager.service - Network Manager
     Loaded: loaded (/usr/lib/systemd/system/NetworkManager.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-06-09 20:06:58 EDT; 5 months 20 days ago
       Docs: man:NetworkManager(8)
   Main PID: 469 (NetworkManager)
      Tasks: 3 (limit: 310)
     Memory: 10.4M
        CPU: 25.811s
     CGroup: /system.slice/NetworkManager.service
             └─469 /usr/sbin/NetworkManager --no-daemon
```

However, on the `ecpix5`, I get this instead:

```
$ dmesg | grep eth0
[  164.846896] liteeth 12001000.mac eth0: irq 2 slots: tx 2 rx 2 size 2048

$ nmcli
eth0: unmanaged
        "eth0"
        ethernet (liteeth), A6:EC:06:4E:EB:63, hw, mtu 1500

lo: unmanaged
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

Use "nmcli device show" to get complete information about known devices and
"nmcli connection show" to get an overview on active connection profiles.

Consult nmcli(1) and nmcli-examples(7) manual pages for complete usage details.

$ systemctl status NetworkManager
● NetworkManager.service - Network Manager
     Loaded: loaded (/usr/lib/systemd/system/NetworkManager.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-06-09 20:18:08 EDT; 17min ago
       Docs: man:NetworkManager(8)
   Main PID: 488 (NetworkManager)
      Tasks: 3 (limit: 310)
     Memory: 3.7M
        CPU: 13.142s
     CGroup: /system.slice/NetworkManager.service
             └─488 /usr/sbin/NetworkManager --no-daemon

Jun 09 20:18:30 fedora-riscv NetworkManager[488]: <info>  [1654820310.3466] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Jun 09 20:18:30 fedora-riscv NetworkManager[488]: <info>  [1654820310.7108] manager: Networking is enabled by state file
Jun 09 20:18:31 fedora-riscv NetworkManager[488]: <info>  [1654820311.2653] dhcp-init: Using DHCP client 'internal'
Jun 09 20:18:31 fedora-riscv NetworkManager[488]: <info>  [1654820311.5099] settings: Loaded settings plugin: keyfile (internal)
Jun 09 20:18:32 fedora-riscv NetworkManager[488]: <info>  [1654820312.8957] settings: Loaded settings plugin: ifcfg-rh ("/usr/lib64/NetworkManager/1.26.8-1.fc33/libnm-settings-plugin-ifcfg-rh.so")
Jun 09 20:18:38 fedora-riscv NetworkManager[488]: <info>  [1654820318.8688] device (lo): carrier: link connected
Jun 09 20:18:39 fedora-riscv NetworkManager[488]: <info>  [1654820319.5663] manager: (lo): new Generic device (/org/freedesktop/NetworkManager/Devices/1)
Jun 09 20:18:41 fedora-riscv NetworkManager[488]: <info>  [1654820321.3213] device (eth0): driver '(null)' does not support carrier detection.
Jun 09 20:18:42 fedora-riscv NetworkManager[488]: <info>  [1654820322.2449] device (eth0): driver 'liteeth' does not support carrier detection.
Jun 09 20:18:43 fedora-riscv NetworkManager[488]: <info>  [1654820323.6352] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
```

The main "highlight" of this is that when the `ecp5rgmii` PHY is in use, the software complains about "carrier detection" being "unsupported". How the software gets to experience this differently is beyond me, as I'm using the exact same kernel with the exact same LiteETH driver, which sees the exact same CSR registers via MMIO, regardless of which board (and PHY) I'm using...

Before I dig into the low-level details of the similarities vs. differences between the two PHY implementations, I'm hoping that someone who's looked at this stuff before can maybe provide some useful starting point(s) as to why this may be happening?

Thanks much in advance!

gsomlo, 2022-12-02:
@enjoy-digital  -- I tried disabling [`with_inband_status`](https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/ecp5rgmii.py#L193) in `ecp5rgmii.py`, but that didn't make a difference.

Beyond that, I haven't (yet?) been able to spot any significant differences beyond just the different IP blocks available for DDR I/O and Delay.

So the only reason I can think of for the different behavior is subtle timing differences in when tx/rx ethernet traffic becomes available via DMA and/or MMIO (csr) registers, since the actual registers and the Linux driver are identical...

If there's anything else I may be missing, please advise :)

rowanG077, 2023-03-10:
@gsomlo A subtle difference is that ECP5 Phy uses loop clocking. See https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/ecp5rgmii.py#L153 and the S7 phy uses an intermediate PLL.

In general this is a bad idea anyway for both cases. It would be better if a fresh 125MHz clock is used for the tx side. From what I gather if both ends of a link use this approach you have a clock recovery loop which means the link will never come up.

gsomlo, 2023-03-10:
@rowanG077 -- thanks for the reply!

This is a problem I ran into with F33, which disappeared when I started using F37. No idea which (intersection of) software component(s) ended up papering over the underlying difference (kernel, systemd, networkmanager, etc).

As such, I'm going to close this issue. Using a dedicated tx clock is probably a good idea, but if we're going to pursue it we should probably open a new issue and/or PR dedicated to that purpose...

Thanks again,
--G

closed 2023-03-10 by gsomlo