enjoy-digital/liteeth #125: Custom RMII PHY PMOD question

machdyne, 2023-01-03:
Hi. I'm working on a custom ethernet PMOD designed specifically to work with LiteX, based on the LAN8720A RMII PHY.

At the moment it doesn't seem to be able to send or receive anything.

I thought it may be related to configuration straps during reset but that didn't seem to be an issue here: https://twitter.com/enjoy_digital/status/1403259359518396418

Any advice for debugging this?

The schematics are available here if anyone is interested: https://github.com/machdyne/katze

Thanks.

```
root@buildroot:~# cat /proc/interrupts 
           CPU0       
  1:         20  SiFive PLIC   2 Edge      eth0
```

```
root@buildroot:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 32:13:3A:20:F3:B8  
          inet addr:192.168.1.199  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::3013:3aff:fe20:f3b8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:20 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:1328 (1.2 KiB)
          Interrupt:1 
```

ozel, 2023-01-21:
the phy's configuration straps could be of concern, depending on the board/pmod wiring. see for example the notes for ulx3s https://github.com/emard/ulx3s-misc/tree/master/examples/eth/rmii/proj and related discussion: https://github.com/emard/ulx3s/issues/14#issuecomment-856644033

since you connected RST, you could make sure that all 3 MODE pins are high while RST is released for configuring auto-negotiation.

did you confirm the ethernet mode with ethtool in linux and compare with the mode detected on the other end?

machdyne, 2023-01-23:
Thanks, this led me to the solution. The problem was the configuration straps. 

ethtool doesn't show anything, maybe because there's no SMI? In any case, I'm now able to send and receive and everything looks good.

I've submitted a PR that provides an option to set the mode during reset.