enjoy-digital/liteeth #147: IP header contains null destination address

jrsa, 2023-09-03:
I instantiated this core in a very simple litex soc design with `add_etherbone` and it seems to work, except that packets sent from the core do not have the destination MAC address set to the one given by the ARP reply, it is set to all zeros. `tcpdump` on the computer can see these packets, but software such as `litex_server` will not. As a workaround, I hardcoded my computer's MAC address [here](https://github.com/enjoy-digital/liteeth/blob/master/liteeth/core/ip.py#L173) and then `litex_server` and `litex_cli` work as expected, except that I would have to resynthesize the design if I needed to switch computers :)

I am fairly new to litex and HDL in general, and thus have not yet tried to debug this myself. I have not discounted the possibility that something is incorrectly configured on my computer that would break ARP but it doesn't seem very likely. For what it's worth, I am able to use ethernet via software, using `--with-ethernet` rather than `--with-etherbone` to build the SoC and booting the prebuilt linux image from [here](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164) with the same host computer.

The target board is LambdaConcept ECPIX-5 which is supported by litex-boards. The command I used to build is
```
python -m litex_boards.targets.lambdaconcept_ecpix5 --with-etherbone --build --load --flash
```

I do not have anything hooked up to the board except a USB cable for the FT2232 interface and an ethernet cable.

tcpdump showing one etherbone packet in each direction, with the reply containing `00:00:00:00:00:00` as the destination address:
```
20:56:14.227301 18:c0:4d:31:67:5c > 10:e2:d5:00:00:00, ethertype IPv4 (0x0800), length 54: (tos 0x0, ttl 64, id 17444, offset 0, flags [DF], proto UDP (17), length 40)
    192.168.1.100.1234 > 192.168.1.50.1234: [bad udp cksum 0x840c -> 0x1288!] UDP, length 12
20:56:14.227350 10:e2:d5:00:00:00 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.1.100 (ff:ff:ff:ff:ff:ff) tell 192.168.1.50, length 46
20:56:14.227353 18:c0:4d:31:67:5c > 10:e2:d5:00:00:00, ethertype ARP (0x0806), length 42: Ethernet (len 6), IPv4 (len 4), Reply 192.168.1.100 is-at 18:c0:4d:31:67:5c, length 28
20:56:14.227433 10:e2:d5:00:00:00 > 00:00:00:00:00:00, ethertype IPv4 (0x0800), length 60: (tos 0x0, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 40)
    192.168.1.50.1234 > 192.168.1.100.1234: [no cksum] UDP, length 12
```

rowanG077, 2023-09-03:
that's definitely not expected. A couple Qs:

- Do you see ARP messages coming in from you FPGA? Wireshark can spot them.
- There recently were some ARP changes. Can you try to use commit df053ae7399a2fc020f72ed7c687f4448497bfe7 that one is prior to the ARP changes.

jrsa, 2023-09-03:
@rowanG077 it is working after reverting to that commit

AndrewD, 2023-10-03:
I can confirm that I'm see the same bahaviour. A simple test case is to build targets/digilent_arty with etherbone.

Pinging etherbone gives the behaviour described above. This seems to be platform specific because it works with ti60_f225.

@enjoy-digital which hardware were you testing with for the arp changes?

enjoy-digital, 2023-10-03:
Thanks all for the feedback, I'll do more tests and will fix it.

enjoy-digital, 2023-10-10:
Exact breaking commit is https://github.com/enjoy-digital/liteeth/commit/c5b53326bb3fc3e9f89f451e9d9663b7a6449dcc.

commit 9d13f612 (2023-10-11): core/arp: Fix missing set of response.mac_address in UPDATE_TABLE and reset update_count at the end of CLEAR state.

Fixes #147.
closed 2023-10-11 by enjoy-digital