enjoy-digital/liteeth #153: How do I add Ethernet support for Xilinx ZCU104?

ratzupaltuff, 2023-11-14:
Hi, i am trying to add Ethernet to the ZCU104 Board.

So far I added the following:
litex_boards/targets/xilinx_zcu104.py:
```
from liteeth.phy.s7rgmii import LiteEthPHYRGMII
...
class BaseSoC(SoCCore):
   ...
   # Ethernet ---------------------------------------------------------------------------------
        if with_ethernet:
            # RGMII Ethernet PHY -------------------------------------------------------------------
            self.ethphy = LiteEthPHYRGMII(
                clock_pads = self.platform.request("eth_clocks"),
                pads       = self.platform.request("eth"))
            self.add_ethernet(phy=self.ethphy)

```
litex_boards/platforms/xilinx_zcu104.py:
```
# Ethernet
    ("eth_clocks", 0,
        Subsignal("tx", Pins("J31")),
        Subsignal("rx", Pins("K31")),
        IOStandard("LVCMOS18")
    ),
    ("eth", 0,
        Subsignal("rx_ctl",  Pins("L30")),
        Subsignal("rx_data", Pins("K32 K33 K34 L29")),
        Subsignal("tx_ctl",  Pins("K30")),
        Subsignal("tx_data", Pins("J32 J34 K28 K29")),
        # Subsignal("rst_n",   Pins("P06")), #dont know what port to use
        Subsignal("mdc",     Pins("L33")),
        Subsignal("mdio",    Pins("L34")),
        # following lines copied from xilinx_ac701
        Misc("SLEW=FAST"),
        Drive(16),
        IOStandard("LVCMOS18"),
    ),
```
But if I run the build I get Errors:
```
...
CRITICAL WARNING: [Netlist 29-336] Instance 'IDDR' of type 'IDDR' failed to retarget to the target device for the following reasons:
 - The SRTYPE attribute is set to 'SYNC'. The target IDDRE1 and ODDRE1 components only support asynchronous reset and thus SRTYPE must be set to 'ASYNC'.
It is suggested to replace the IDDR component with the IDDRE1 component and update the design to the functionality and capabilities of the new output IDDR register or else modify the IDDR instantiation to match the above characteristics.
CRITICAL WARNING: [Netlist 29-336] Instance 'IDDR_1' of type 'IDDR' failed to retarget to the target device for the following reasons:
...
CRITICAL WARNING: [Netlist 29-335] Instance 'IDELAYE2' of type 'IDELAYE2' may have failed to retarget to the target device for the following reasons:
 - The IDELAY_VALUE attribute is set to 26 - The target architecture IDELAYE3 component does not support delay values greater than 16.
It is suggested to replace the component with the native IDELAYE3 component to update the design to the functionality and capabilities of the target architecture DELAY cell or else modify the instantiation to match the above characteristics. Please consult UG 1026: UltraScale Architecture Migration User Guide as well as UG 571: UltraScale SelectIO User Guide and the appropriate datasheet for re-targeting and other details about the IDELAYE3 and ODELAYE3 components.
...
CRITICAL WARNING: [Netlist 29-186]  Instance 'ODDR' of type 'ODDR' failed to retarget to the target device for the following reasons:
 - The SRTYPE attribute is set to 'SYNC'. The target IDDRE1 and ODDRE1 components only support asynchronous reset and thus SRTYPE must be set to 'ASYNC'.
It is suggested to replace the ODDR component with the native ODDRE1 component to update the design to the functionality and capabilities of the target architecture output DDR register or else modify the ODDR instantiation to match the above characteristics.
...
Parsing XDC File [/home/XXX/build/zcu104/gateware/zcu104.xdc]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Site location (PS8_X0Y0) is not valid for the shape with the following elements: 
OBUF
eth_clocks_tx
 [/home/XXX/build/zcu104/gateware/zcu104.xdc:769]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Site location (PS8_X0Y0) is not valid for the shape with the following elements: 
IBUF/INBUF_INST
eth_clocks_rx
IBUF/IBUFCTRL_INST
 [/home/XXX/build/zcu104/gateware/zcu104.xdc:773]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Site location (PS8_X0Y0) is not valid for the shape with the following elements: 
IBUF_1/INBUF_INST
eth_rx_ctl
IBUF_1/IBUFCTRL_INST
 [/home/XXX/build/zcu104/gateware/zcu104.xdc:777]
WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/XXX/build/zcu104/gateware/zcu104.xdc:778]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/XXX/build/zcu104/gateware/zcu104.xdc:779]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Site location (PS8_X0Y0) is not valid for the shape with the following elements: 
IBUF_2/INBUF_INST
eth_rx_data[0]
IBUF_2/IBUFCTRL_INST
...
CRITICAL WARNING: [Project 1-560] Could not resolve non-primitive black box cell 'IDDR_BLACKBOX' instantiated as 'IDDR'. 5 instances of this cell are unresolved black boxes. [/home/XXX/build/zcu104/gateware/zcu104.v:28253]
CRITICAL WARNING: [Project 1-560] Could not resolve non-primitive black box cell 'IDELAYE2_BLACKBOX' instantiated as 'IDELAYE2'. 5 instances of this cell are unresolved black boxes. [/home/XXX/build/zcu104/gateware/zcu104.v:28241]
CRITICAL WARNING: [Project 1-560] Could not resolve non-primitive black box cell 'ODDR_BLACKBOX' instantiated as 'ODDR'. 6 instances of this cell are unresolved black boxes. [/home/XXX/build/zcu104/gateware/zcu104.v:28132]
...
6 Infos, 0 Warnings, 0 Critical Warnings and 17 Errors encountered.
opt_design failed
ERROR: [Common 17-39] 'opt_design' failed due to earlier errors.
```

When I rename
ODDR -> ODDRE1
IDELAYE2 -> IDELAYE3
IDDR -> IDDRE1
in liteeth/phy/s7rgmii.py I get a different error:
```
...
WARNING: [Synth 8-7023] instance 'ODELAYE3_46' of module 'ODELAYE3' has 13 connections declared, but only 7 given [/home/XXX/build/zcu104/gateware/zcu104.v:23128]
WARNING: [Synth 8-7023] instance 'IDELAYE3_3' of module 'IDELAYE3' has 14 connections declared, but only 7 given [/home/XXX/build/zcu104/gateware/zcu104.v:23149]
WARNING: [Synth 8-7023] instance 'ISERDESE3_4' of module 'ISERDESE3' has 10 connections declared, but only 7 given [/home/XXX/build/zcu104/gateware/zcu104.v:23188]
INFO: [Common 17-14] Message 'Synth 8-7023' appears 100 times and further instances of the messages will be disabled. Use the Tcl command set_msg_config to change the current settings.
INFO: [Synth 8-6157] synthesizing module 'IBUF' [/tools/Xilinx/Vivado/2022.2/scripts/rt/data/unisim_comp.v:71326]
INFO: [Synth 8-6155] done synthesizing module 'IBUF' (0#1) [/tools/Xilinx/Vivado/2022.2/scripts/rt/data/unisim_comp.v:71326]
ERROR: [Synth 8-439] module 'PDDRE1' not found [/home/XXX/build/zcu104/gateware/zcu104.v:28132]
        Parameter DDR_CLK_EDGE bound to: SAME_EDGE - type: string 
ERROR: [Synth 8-6156] failed synthesizing module 'zcu104' [/home/XXX/build/zcu104/gateware/zcu104.v:21]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:07 ; elapsed = 00:00:07 . Memory (MB): peak = 2665.012 ; gain = 353.527 ; free physical = 985 ; free virtual = 8647
Synthesis current peak Physical Memory [PSS] (MB): peak = 2274.151; parent = 2020.821; children = 253.406
Synthesis current peak Virtual Memory [VSS] (MB): peak = 3669.113; parent = 2665.016; children = 1004.098
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
55 Infos, 215 Warnings, 0 Critical Warnings and 3 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
...
```

I am new to Litex and FPGA programming with Vivado. Can someone help me, what the error means and what needs to be done to resolve it? Is there another liteeth-phy that is working with the zcu104?

rowanG077, 2023-11-14:
I'm not experienced with Xilinx but from looking at their site the ZCU104 is an UltraScale+ part. So it seems you require the [usrgmii Phy](https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/usrgmii.py) not the `s7rgmii` you used.

ratzupaltuff, 2023-11-15:
Hi, Thank you for your answer! I overlooked the usrgmii Phy completely, my bad. I used it now and adapted my files a bit:

litex_boards/targets/xilinx_zcu104.py:
```
from liteeth.phy.usrgmii import LiteEthPHYRGMII
...
# Ethernet ---------------------------------------------------------------------------------
        if with_ethernet:
            # RGMII Ethernet PHY -------------------------------------------------------------------
            self.ethphy = LiteEthPHYRGMII(
                clock_pads = self.platform.request("eth_clocks"),
                pads       = self.platform.request("eth"),
                rx_delay=1e-9,
                tx_delay=1e-9,
                usp=True)
            
            # Change ref clk for IDELAYE3 - Copied from avnet_aesku40
            for special in self.ethphy.rx._fragment.specials:
                if special.name_override == "IDELAYE3":
                    for item in special.items:
                        if item.name == "REFCLK_FREQUENCY":
                            item.value=500.00

            self.add_ethernet(phy=self.ethphy)
```
litex_boards/platforms/xilinx_zcu104.py:
```
# Ethernet
    ("eth_clocks", 0,
        Subsignal("tx", Pins("J31")),
        Subsignal("rx", Pins("K31")),
        IOStandard("LVCMOS18")
    ),
    ("eth", 0,
        Subsignal("rx_ctl",  Pins("L30")),
        Subsignal("rx_data", Pins("K32 K33 K34 L29")),
        Subsignal("tx_ctl",  Pins("K30")),
        Subsignal("tx_data", Pins("J32 J34 K28 K29")),
        # Subsignal("rst_n",   Pins("P06")), #dont know what port to use
        Subsignal("mdc",     Pins("L33")),
        Subsignal("mdio",    Pins("L34")),
        IOStandard("LVCMOS18"),
    ),
```
I get the following errors:

```
Start Applying XDC Timing Constraints
---------------------------------------------------------------------------------
CRITICAL WARNING: [Synth 8-3321] create_clock attempting to set clock on an unknown port/pin for constraint at line 841 of /home/XXX/build/zcu104/gateware/zcu104.xdc. [/home/XXX/build/zcu104/gateware/zcu104.xdc:841]
CRITICAL WARNING: [Synth 8-3321] create_clock attempting to set clock on an unknown port/pin for constraint at line 843 of /home/XXX/build/zcu104/gateware/zcu104.xdc. [/home/XXX/build/zcu104/gateware/zcu104.xdc:843]
...
Parsing XDC File [/home/XXX/build/zcu104/gateware/zcu104.xdc]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Could not legally place terminal eth_clocks_tx at N29 (PS8_X0Y0) since it belongs to a shape containing instance ODDRE1. The shape requires relative placement between eth_clocks_tx and ODDRE1 that can not be honoured because it would result in an invalid location for ODDRE1. [/home/XXX/build/zcu104/gateware/zcu104.xdc:769]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Site location (PS8_X0Y0) is not valid for the shape with the following elements: 
IBUF/INBUF_INST
eth_clocks_rx
IBUF/IBUFCTRL_INST
 [/home/XXX/build/zcu104/gateware/zcu104.xdc:773]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Could not legally place terminal eth_rx_ctl at N29 (PS8_X0Y0) since it belongs to a shape containing instance IDDRE1. The shape requires relative placement between eth_rx_ctl and IDDRE1 that can not be honoured because it would result in an invalid location for IDDRE1. [/home/XXX/build/zcu104/gateware/zcu104.xdc:777]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Could not legally place terminal eth_rx_data[0] at N29 (PS8_X0Y0) since it belongs to a shape containing instance IDDRE1_1. The shape requires relative placement between eth_rx_data[0] and IDDRE1_1 that can not be honoured because it would result in an invalid location for IDDRE1_1. [/home/XXX/build/zcu104/gateware/zcu104.xdc:781]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Could not legally place terminal eth_rx_data[1] at N29 (PS8_X0Y0) since it belongs to a shape containing instance IDDRE1_2. The shape requires relative placement between eth_rx_data[1] and IDDRE1_2 that can not be honoured because it would result in an invalid location for IDDRE1_2. [/home/XXX/build/zcu104/gateware/zcu104.xdc:785]
...
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Site location (PS8_X0Y0) is not valid for the shape with the following elements: 
eth_mdc_OBUF_inst
eth_mdc
 [/home/XXX/build/zcu104/gateware/zcu104.xdc:817]
CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Site location (PS8_X0Y0) is not valid for the shape with the following elements: 
eth_mdio_IOBUF_inst/INBUF_INST
eth_mdio_IOBUF_inst/OBUFT
eth_mdio
eth_mdio_IOBUF_inst/IBUFCTRL_INST
 [/home/XXX/build/zcu104/gateware/zcu104.xdc:821]
...
ERROR: [DRC UCIO-1] Unconstrained Logical Port: 14 out of 134 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this c [... cut ...]
ERROR: [Vivado 12-1345] Error(s) found during DRC. Bitgen not run.
INFO: [Common 17-83] Releasing license: Implementation
6 Infos, 19 Warnings, 0 Critical Warnings and 2 Errors encountered.
write_bitstream failed
write_bitstream: Time (s): cpu = 00:00:13 ; elapsed = 00:00:07 . Memory (MB): peak = 4465.363 ; gain = 0.000 ; free physical = 449 ; free virtual = 8185
ERROR: [Common 17-39] 'write_bitstream' failed due to earlier errors.
```
Can someone give me a hint what the problem may be or even how to solve it?

rowanG077, 2023-11-15:
It seems it not taking the pin locations constrained properly. Does litex generate the proper XDC file here? Did you load it into the tooling properly? Are you sure the locations you specified are good? 

Beyond this I can't really help.

trabucayre, 2023-11-15:
ZCU 104 is a zynqmp SoC Ultrascale+ (ie FPGA+CPU).
Ethernet port is connected to PS (CPU) side. PL (FPGA) can't access to these pins.

ratzupaltuff, 2023-11-15:
Ah, thank you! I think I missed that in the manual. I will use a different board then.

closed 2023-11-15 by ratzupaltuff