enjoy-digital/liteeth #155: Unable to simulate generated liteeth core

ameetgohil, 2024-02-15:
Both Vivado sim and icarus are unable get past time 0. I get the following error in vivado
**FATAL_ERROR: Iteration limit 10000 is reached. Possible zero delay oscillation detected where simulation time can not advance. Please check your source code. Note that the iteration limit can be changed using switch -maxdeltaid.
Time: 0 ps  Iteration: 10000**

Gen yml file: 
```yml
# PHY ----------------------------------------------------------------------
phy:        LiteEthPHYRMII
vendor:     xilinx
toolchain:  vivado
# Core ---------------------------------------------------------------------
clk_freq:   50e6
core:       udp
data_width: 32
endianness: big

udp_ports: {
  "udp0": {
    "data_width": 32,
    "tx_fifo_depth": 1024,
    "rx_fifo_depth": 1024,
  },
  "udp1": {
    "data_width": 32,
    "tx_fifo_depth": 1024,
    "rx_fifo_depth": 1024,
  },
}
```

Minimal testbench to reproduce to the issue:
```systemverilog
module tb;

logic clk, rst;
liteeth_core dut
(.rmii_clocks_ref_clk(clk),
    .sys_clock(clk),
.sys_reset(rst));

initial begin
    clk = 0;
    forever #10ns clk = ~clk;
end

initial begin
    rst = 1;
    repeat(10) @(posedge clk);
    rst = 0;
    repeat(10000) @(posedge clk);
end

endmodule
```

It seems like both icarus and vivado simulators have an issue with the following lines of code. 
```verilog
 if (mac_packetizer_is_ongoing2) begin
        mac_packetizer_sink_ready <= (~mac_packetizer_sink_valid);
  end
```
```verilog
if (arp_tx_packetizer_is_ongoing2) begin
     arp_tx_packetizer_sink_ready <= (~arp_tx_packetizer_sink_valid);
 end
```
```verilog
if (ip_tx_packetizer_is_ongoing2) begin
     ip_tx_packetizer_sink_ready <= (~ip_tx_packetizer_sink_valid);
end
````
```verilog
 if (icmp_tx_packetizer_is_ongoing2) begin
      icmp_tx_packetizer_sink_ready <= (~icmp_tx_packetizer_sink_valid);
  end
```
```verilog
if (tx_packetizer_is_ongoing2) begin
          tx_packetizer_sink_ready <= (~tx_packetizer_sink_valid);
end
```
If you comment these lines, the simulator starts advancing again.

rowanG077, 2024-02-15:
I think this is the same issue reported here: https://github.com/enjoy-digital/litex/issues/1841

See https://github.com/steveicarus/iverilog/issues/1028 for some background on why it doesn't work.

ameetgohil, 2024-02-15:
Thanks! I used a sed command to replace all the `always @(*)` to `always_comb` and switched file extension to `.sv` 
With this, Vivado sim was able resolve the signal being set and read in the same process.
`sed -i 's/always @(\*)/always_comb/g' filename`

closed 2024-02-15 by ameetgohil