enjoy-digital/liteeth #159: Updating `liteeth` PyPI package?

fischermoseley, 2024-04-09:
Hello! Just wanted to say thank you for putting together this package - using it has been genuinely magical.
 
I wanted to ask if it'd be possible to update the version on PyPI to something newer than 2022.12. I'm currently working on a project that uses LIteeth 2023.12, but I'm trying to run it inside pyodide, whose builtin `micropip` doesn't support dependency resolution for non-PyPI packages. See below:

https://pyodide.org/en/stable/usage/loading-packages.html#micropip

As a result I'm not able to load LiteEth 2023.12 via git - and I can only get 2022.12 via PyPI, which isn't compatible with my project.

I'm also happy to make a PR for a Github Action that automatically pushes to the PyPI lists when a new release is published, if that'd be useful.

Thank you!
-Fischer

enjoy-digital, 2024-04-14:
Hi @fischermoseley,

thanks for the feedback. We indeed have to automate pushing release to PyPI. The initial PyPi packages have been contributed by @timkpaine and I've not yet been able to continue this effort, I'll try spend more time on this but any help is welcome!

fischermoseley, 2024-04-14:
Thank you both! Happy to help with automating this if that would be welcome/nonbreaking for existing users. And happy to extend that to the rest of the LiteX cores as well if that would make things easier to distribute.

timkpaine, 2024-04-14:
That's the plan, I got the first 5 or so metadata and format ready but there's the full set that needs to be completed before the next stage in automating uploads can be done.

Peter-van-Tol, 2024-07-01:
I think this should be possible to be automated using GitHub actions. The dependency-graph for `liteeth` is: `litex` -> `liteiclink` -> `liteeth`. In order to have this package automated for release to  pypi.org, these dependencies should also be automated.

If wished for, I can produce a 'publish to PyPi' GitHub action....

timkpaine, 2024-07-01:
Some of the libraries still need to be fixed to be publishable, they have invalid setup.py

Peter-van-Tol, 2024-07-02:
Drom the three libraries mentioned above, which ones are still invalid?

When making them valid, is there a wish to move away from setup.py and go for a more modern alternative (pyproject.toml)?

enjoy-digital, 2024-07-02:
Hi,

as a first step, we can indeed update the repos that don't have a valid setup.py (@timkpaine: Could you do the list?) We could then see if we switch to an alternative (but if so, I need to evaluate the impact and we'll probably do it for all repos directly).

Peter-van-Tol, 2024-07-02:
Sounds like a good plan. Bsed on the list, I would be more then willing to create PR's. My project LitexCNC would become much easier to use when the updates of LiteX are available on PyPi. There is mutually beneficent.

fischermoseley, 2024-07-02:
Thanks all! I'm also more than happy to make PRs here (my project would also greatly benefit from having LiteX on PyPI). I imagine the order those PRs are merged in will be important as well to make sure there aren't unmet dependencies in downstream packages like `liteeth`. @timkpaine, could you also include this order in your list?