enjoy-digital/liteeth #162: ntxslots=1 is broken

sensille, 2024-06-29:
When I try to build for example litex_boards/targets/linsn_rv901t.py with --with-ethernet, but add ntxslots=1 to the add_ethernet line, build fails with:

> Traceback (most recent call last):
>   File "/home/me/litex-boards/litex_boards/targets/./linsn_rv901t.py", line 122, in <module>
>     main()
>   File "/home/me/litex-boards/litex_boards/targets/./linsn_rv901t.py", line 115, in main
>     builder.build(**parser.toolchain_argdict)
>   File "/home/me/litex/litex/soc/integration/builder.py", line 415, in build
>     vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
>           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/soc/integration/soc.py", line 1498, in build
>     return self.platform.build(self, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/build/xilinx/platform.py", line 98, in build
>     return self.toolchain.build(self, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/build/xilinx/ise.py", line 57, in build
>     return GenericToolchain.build(self, platform, fragment, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/build/generic_toolchain.py", line 86, in build
>     v_output = platform.get_verilog(self.fragment, name=build_name, **kwargs)
>                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/build/xilinx/platform.py", line 88, in get_verilog
>     return GenericPlatform.get_verilog(self, *args,
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/build/generic_platform.py", line 472, in get_verilog
>     return verilog.convert(fragment, platform=self, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 669, in convert
>     verilog += _generate_combinatorial_logic_synth(f, ns)
>                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 514, in _generate_combinatorial_logic_synth
>     r += _generate_node(ns, AssignType.NON_BLOCKING, 1, g[1])
>          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 327, in _generate_node
>     return "".join(_generate_node(ns, at, level, n, target_filter) for n in node)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 327, in <genexpr>
>     return "".join(_generate_node(ns, at, level, n, target_filter) for n in node)
>                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 323, in _generate_node
>     return _tab*level + _generate_expression(ns, node.l)[0] + assignment + _generate_expression(ns, node.r)[0] + ";\n"
>                                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 281, in _generate_expression
>     return _generate_operator(ns, node)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 217, in _generate_operator
>     r1, s1 = _generate_expression(ns, operands[0])
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 285, in _generate_expression
>     return _generate_slice(ns, node)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/me/litex/litex/gen/fhdl/verilog.py", line 248, in _generate_slice
>     assert (node.stop - node.start) >= 1
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> AssertionError

I use this parameter in my litehm2 project to save some resources.

enjoy-digital, 2024-07-02:
Hi @sensille,

thanks for the feedback, I indeed reproduce the issue and will fix it.

enjoy-digital, 2024-07-02:
This should be fixed with https://github.com/enjoy-digital/liteeth/commit/ec7320f00395012befec096caa7b1823f6b174f6.

sensille, 2024-07-13:
Builds again, thanks!

closed 2024-07-13 by sensille